opulo-inc/lumenpnp #718: Error while reading machine.xml in new configuration v3.2-01

kheygetyan, 2024-02-19:
Uploaded [LumenPnP-Config-v3.2-01.zip](https://github.com/opulo-inc/lumenpnp/releases/download/v3.2.0/LumenPnP-Config-v3.2-01.zip) to a clean folder (.openpnp2)
And the OpenPnp gives an error when launched
[image]

sphawes, 2024-02-29:
There's a new version of the config in the V3.2 release now! Does that fix the issue?

atanisoft, 2024-02-29:
This is likely an issue due to difference in version of OpenPnP. Which version are you using and which is @sphawes using for the release config?

kheygetyan, 2024-02-29:
> This is likely an issue due to difference in version of OpenPnP. Which version are you using and which is @sphawes using for the release config?

I'm using OpenPnP 2.0

atanisoft, 2024-02-29:
Can you cross check the version in the about box? I'd expect a date / time stamp on it and not just 2.0 (which is unfortunately what they almost all are labeled)

kheygetyan, 2024-03-01:
> Can you cross check the version in the about box? I'd expect a date / time stamp on it and not just 2.0 (which is unfortunately what they almost all are labeled)

OpenPnP Version: 2023-03-15_00-30-21.460d8aa

atanisoft, 2024-03-03:
@kheygetyan Please upgrade your OpenPnP build to the 2023-04-05 release as linked from https://docs.opulo.io/openpnp/install/.

kheygetyan, 2024-03-04:
> @kheygetyan Please upgrade your OpenPnP build to the 2023-04-05 release as linked from https://docs.opulo.io/openpnp/install/.

Thank you!
With the Test version of the OpenPnp now starts without errors

closed 2024-03-04 by kheygetyan