opulo-inc/lumenpnp #727: Pneumatics plumbing inconsistencies

asm308, 2024-05-31:
# Version Number
Branch rev05-mobo

# Bugfix or Enhancement
Bug or I don't understand how its going to work

# Description
I've been following along with the modifications for the rev05 mobo.  In particular the new enclosure (which is a much nicer solution than the staging plate).  I appreciate that this is a work in progress

I think I see an inconsistency or maybe I just don't understand how its supposed to connect up.  The toolheads take a 4mm tube which would come from the output side of the enclosure.  Internally the pump connects to the valve using 6mm tube and then the 6mm tube continues to the tee reducer.  A 4mm needs to go to the vacuum sensor on the mobo and a 4mm needs to go out

The Tee reducer in the design is a 6-4-6 which would give a 4mm to the vacuum sensor but that would only leave a 6mm output to the toolhead which requires a 4mm tube

Seems you would need a 4-6-4 tee but I can't find one of those anywhere 

# Suggested Solution
Please describe a suggested solution, if you have one in mind.

atanisoft, 2024-05-31:
> Seems you would need a 4-6-4 tee but I can't find one of those anywhere

The design has (since the beginning) used a 4-6-4 Y fitting. The BOM calls for [this](https://www.automationdirect.com/adc/shopping/catalog/pneumatic_components/push-to-connect_union_pneumatic_fittings_(thermoplastic)/union_y_reducer/ury6m-4m) one, another location to find them is Amazon such as [this](https://www.amazon.com/dp/B07C7LX5N2?ref=ppx_yo2ov_dt_b_product_details&th=1) one (which I'm using on my heavily modified Lumen)

asm308, 2024-05-31:
Yes it currently calls for a Y fitting but the new enclosure calls for a Tee fitting.

atanisoft, 2024-05-31:
Ahh, I missed it was a tee fitting. https://www.fixsupply.com/zusa-tf-ptc-555-push-to-connect-tube-fitting-nylon-plastic-reducing-union-tee-6mm-x-4mm-tube-od looks like the right fitting.

asm308, 2024-05-31:
That fitting has 2 6mm sockets and one 4mm.  BUT if i understand correctly, the design calls for 2x4mm and 1 x 6mm.  The 6mm comes from the valve and both the vacuum sensor and tool head need 4mm pipes.  I couldn't find one that did 2 x 4mm and 1x 6mm

atanisoft, 2024-05-31:
Ahh, you are right. I missed that on the listing. I'm sure there is a listing for it somewhere.

asm308, 2024-05-31:
well i couldn't find one anywhere only a 6-4-6 or 6-6-6 or 4-4-4 but not a 6-4-6.  Not sure why when you can get Y shaped ones with2 x4 and 1x6

asm308, 2024-05-31:
Found one in US
https://www.fixsupply.com/zusa-tf-ptc-554-push-to-connect-tube-fitting-nylon-plastic-reducing-union-tee-6mm-x-4mm-tube-od
Now just need to find uk supplier.  Thanks for the help

lucian151, 2024-06-11:
You two are on the right track, these are hard for us to source right now as well!

closed 2024-07-12 by lucian151