opulo-inc/lumenpnp #733: kicad projects reference private libraries not in the github repository

asm308, 2024-08-31:
# Version Number
rev05 branch

# Bugfix or Enhancement
Incomplete design assets

# Description
The following pcb designs in kicad reference libraries that are private ones to stephens user id.
x-y limit switch two references to step files for the limit switches
**Errors were encountered loading footprints:
Footprint library '/Users/stephen/lumenpnp/lib/kicad/index.pretty' not found.** 
[image]

There are also a couple of local references in the mobo design.

I haven't checked all of the PCB designs

There are also a lot of references to a local library e.g.
model "${KIPRJMOD}/../../../lib/kicad/index.3dshapes/Littelfuse-Nano2-154.wrl"

List of references in mobo design:
Line   2213: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B2B-PH-SM4-TB.stp"
	Line   3079: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B4B-PH-SM4-TB.stp"
	Line   6679: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B2B-PH-SM4-TB.stp"
	Line   7560: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B4B-PH-SM4-TB.stp"
	Line   8558: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B2B-PH-SM4-TB.stp"
	Line  13608: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B4B-PH-SM4-TB.stp"
	Line  15233: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B5B-PH-SM4-TB.stp"
	Line  18639: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B6B-PH-SM4-TB.stp"
	Line  22361: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B4B-PH-SM4-TB.stp"
	Line  30333: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B4B-PH-SM4-TB.stp"
	Line  34397: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B4B-PH-SM4-TB.stp"
	Line  39068: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B4B-PH-SM4-TB.stp"
	Line  44000: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B2B-PH-SM4-TB.stp"
	Line  77103: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B3B-PH-SM4-TB.stp"
	Line  78589: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B2B-PH-SM4-TB.stp"
	Line  85760: 		(model "${KIPRJMOD}/../../../lib/kicad/index.3dshapes/Littelfuse-Nano2-154.wrl"
	Line  86558: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B4B-PH-SM4-TB.stp"
	Line  97592: 		(model "${KIPRJMOD}/../../../lib/kicad/jst-ph.3dshapes/B3B-PH-SM4-TB.stp"

Which I think means we don't have those either 

Opening the mobo project pcd, the first thing we see is 
[image]

If I look at the 3d view of the circuit board I don't see any of the connectors for example.  eg
[image]

# Suggested Solution
Publish the footprint/3D assets which are not in standard libraries in github so that they can be downloaded and added to users configurations of kicad.  It would also be useful to have a brief description on how to set up kicad to reference these resources

atanisoft, 2024-08-31:
The references are present in the repository but the links appear to be incorrect, the files live in https://github.com/opulo-inc/lumenpnp/tree/rev05-mobo/lib/kicad.

Though the one reference with @sphawes name in it, that should be converted over to using `${KIPRJMOD}` based reference links.

asm308, 2024-09-01:
Thanks for the update Mike, I hadn't appreciated the assets were already there.  There are two stephen references in the x-y limit switch pcb as there are two different switch types.
I'll download the kicad library - thanks again