opulo-inc/lumenpnp #736: TAG Connect SMD "Connector" for Programming

Josh-B-1209, 2024-12-18:
# Version Number
Latest (V4)

# Enhancement
Potential enhancement

# Description
Noticed particularly with feeder pcb's that a physical connector is used for programming. I recommend using tag connect, Website: https://www.tag-connect.com/, it is essentially an smd footprint comprising of 6, 10 or 14 circular smd pads arranged in a 2 by X rectangle. 

The idea is that since the feeder or even the main pick n place motherboard microcontroller would only need to be programmed once, it doesn't make sense to use a physical connector like a jst for example considering it adds to the BOM cost when making a large number of boards.

Tag connect sells cables which on 1 end has an arrangement of pogo pins and 3 alignment pins which will correspond to the smd pads and small alignment holes on the pcb. This allows a pcb to be programmed by a user without needing to first solder on a connector or... automatically with a pick and place machine with a custom toolhead (: .

Voltlog Video Explanation and Demonstration: https://youtu.be/9-e_gcYaEM0?si=SZOb3TbV-59RbMbt&t=158

(It should be noted that Tag Connect footprints are included in Kicad's footprint and symbol library by default, no need to download or create.)

# Suggested Solution
Replace JTAG Headers with Tag Connect SMD Pads

daveismith, 2024-12-18:
Personally I'm not the biggest fan of the tag connect having used them in other projects. In fact it came up as undesirable in an hardware review for a team I'm part of recently with the firmware / software folks not being fans but the HW folks thinking it was a great idea. The particular complaints were that connecting / disconnecting the tag connect cable (with retention clips) is cumbersome and that the cables are significantly more expensive than standard 1.27mm cables / connectors. 

For manufacturing in small quantities it's a reasonable solution which doesn't add board cost. Once you get to medium or large scale production, you'll likely be using a fixture with pogo pins that would replace the tag connect so it doesn't give a lot of benefit. A fixture of this style can either hit testpoints or the pads where you would typically populate the connector.

Having said all of this, I understand that this is going to be an item of personal preference. If there's a footprint which can allow both tag connect & soldered connectors it may provide a middle ground which could make both those who like and dislike the tag-connect system happy.

JordyDH, 2025-01-14:
You could also look into the SKEDD system of wurth.
https://www.we-online.com/en/components/products/REDFIT_IDC_SKEDD
Connectors are cheap and easy to self make, and they have released a footprint for programming and debug purpose.
And you "can" use probes to directly interface with the SKEDD, but for this I would recommend a dedicated test point to connect to.